ut-robotics/picr22-team-snax #4: [Mechanics] underdefined sketches

Akustav, 2022-10-01:
There are currently under-defined sketches:
[image]

This can lead to unpredictable changes in model features when another sketch is changed. You should add dimensions until sketches turn black.

ToxaRyd, 2022-10-04:
Did my best but errors of over constrain occur...

Akustav, 2022-10-09:
Hm, currently seems fine to me. Overall when it gives you over-defined warnings, it means that you added multiple different dimensions to the same area. Usually the simplest way to see what is undefined is to take the sketch and try dragging some blue lines around. This shows you what elements start moving.

closed 2022-10-09 by Akustav