ut-robotics/picr22-team-snax #6: [Mechanics] Thrower and ramp are not centered

Akustav, 2022-10-01:
Is this intended, that the thrower and internal ramp don't have the same center point? The further the ball travels in the thrower, the less it is accelerated by the motor.

[image]

ToxaRyd, 2022-10-04:
Yep, I was suggested that the first contact with the ball should have the motor, and only after the ramps

Akustav, 2022-10-09:
Hm okay... I would suggest to reduce the delta a little, currently it is around 9.5mm:

[image]

You might remove acceleration potential by reducing contact pressure so fast.

ToxaRyd, 2022-10-14:
Have decreased this distance by 3mm if I am not mistaken....resolved

closed 2022-10-27 by Akustav