ut-robotics/picr22-team-snax #14: Don't use metric grid on the schematic

ReikoR, 2022-11-11:
Industry standard unfortunately is 100 mil grid. All the symbols are designed for it. Using different grid can cause issues with things not aligning with each other.

On the PCB it's better to use metric grid.

ReikoR, 2022-11-22:
[image]

There are a lot of "Off grid" warnings:
[image]

Use 100 mil grid and place all schematic object to that grid.

You can set grid to 100 mil (toggle grid with G), select objects on the schematic, use Edit -> Align -> Align To Grid (Shift + Ctrl + D) and adjust manually as necessary.

If symbols were not created with 100 mil grid, then the symbols also need to be changed.

ReikoR, 2022-11-29:
Move schematic objects to 100 mil grid.

One part of the schematic with misalignement:
[image]

markus-marandi, 2022-12-02:
done

ReikoR, 2022-12-03:
[image]
[image]
[image]
[image]

markus-marandi, 2022-12-03:
fixed

ReikoR, 2022-12-04:
[image]
[image]
[image]

markus-marandi, 2022-12-04:
[image]
[image]
[image]

markus-marandi, 2022-12-05:
[image]
[image]
[image]
[image]
[image]

closed 2022-12-05 by ReikoR