ut-robotics/picr22-team-snax #30: Use switching regulator instead of linear regulator for servo power

ReikoR, 2022-11-22:
[image]
[image]

Linear regulators are too inefficient for higher currents.

We have these components available in the lab:

- XTPS62933DRLR (TPS62933 buck converter)
- SRP6060FA-5R6M (Inductor 5.6uH 10A)

You can use TI WEBENCH to get recommended designs for TI components:
https://webench.ti.com/power-designer/switching-regulator

**Considering that there is not too much time left to complete the design, I would recommend not adding servo connection and continuing with a simpler design.**

markus-marandi, 2022-12-02:
removed servo NCV and 215079-4 components

ReikoR, 2022-12-03:
If you're not using servos, then remove switching regulator and use linear regulator (NCV8730).

Isolators don't need high current and layout of NCV8730 is much simpler.

markus-marandi, 2022-12-03:
added ncv8730

ReikoR, 2022-12-04:
There is no 3.3V regulator any more.
And switching regulator (TPS62933) is still there.

markus-marandi, 2022-12-04:
removed TPS62933 and added ncp164
[image]

markus-marandi, 2022-12-04:
[image]

ReikoR, 2022-12-05:
[image]

Add more input and output capacitance to NCP164.

From [NCP164 datasheet](https://www.onsemi.com/pdf/datasheet/ncp164-d.pdf):
> **Input Decoupling (C<sub>IN</sub>)**
> It is recommended to connect at least 1 uF ceramic X5R or X7R capacitor between IN and GND pin of the device.

> **Output Decoupling (C<sub>OUT</sub>)**
> The device is designed to be stable with standard ceramics capacitors with values of 1 uF or greater.

closed 2022-12-05 by ReikoR