ut-robotics/picr22-team-snax #32: Wider taper at the front of the robot

Akustav, 2022-11-26:
I suggest having a wider taper at the front of the robot. This reduces the required accuracy for aiming the ball. Also dont forget that the ball is round so the tapered front should make contact with the ball from around its center, else the ball might make a yanky motion, when it reaches the thrower

[image]

ToxaRyd, 2022-11-26:
Yep I was thinking about that so made those wider, the thing I was concerned about is that the ball may be pressed early, towards the wall..but I will make it as you has shown

ToxaRyd, 2022-11-26:
Seems to be done, waiting for approval

closed 2022-11-27 by Akustav