ut-robotics/picr22-team-snax #34: More support for camera legs

Akustav, 2022-11-26:
I suggest a little more support for the camera legs since they can wobble about, when the robot is is motion. The slots do not help. Maybe add a plate to the middle of the legs?

[image]

ToxaRyd, 2022-11-26:
Yeah, I planned to add there a plane, I just have not finished yet)

Akustav, 2022-11-27:
I see something added. How do you plan to fasten it. Additionally how do you plan to manufacture it?

[image]

ToxaRyd, 2022-11-27:
3d printing, cause it is just rigid component, no fixing needed, maybe just position fix through the adjustment of the holes in the camera legs

Akustav, 2022-11-27:
Some force needs to hold it in place, no? If it is just rattling there, there is little rigidity and support it is adding. If it were somewhat wider and was squeezed between the two legs, then maybe it can be a friction fit.

ToxaRyd, 2022-11-28:
please check the newer version

closed 2022-11-30 by Akustav