ut-robotics/picr22-team-snax #36: Underdefined sketch

Akustav, 2022-11-26:
The sketch is underdefined for the camera legs.

[image]

Also... This camera leg reminds me an awful lot of the mount used by this robot. whats up with that:

[image]

ToxaRyd, 2022-11-27:
defining sketches fully is a highly time consuming habit especially when smth needs to be redone, also I have never experienced any problems with that throughout 6 years of CAD exp...

ToxaRyd, 2022-11-27:
Speaking of those mount slots, I may redo the entire leg to be thinner at those places by reducing the mount slots...

Akustav, 2022-11-27:
Fully defining sketches is a good practice. Currently this sketch looks like a carbon copy-paste from the previous years robot that borders on plagiarism with its distinct features.

A side-to-side here:

[image]

ToxaRyd, 2022-11-28:
Indeed the initial sketch was copy pasted from the test model given to my team, further - modified much. The reason - is simply saving time in terms of defining all the dimensions and of course, saving all angles to eliminate any extra job to be done for my team, if to speak about those mounting holes - I did find them very useful when additional electronics need to be fixed somewhere...

closed 2022-11-30 by Akustav