ut-robotics/picr22-team-snax #37: Cameras RGB sensor is not centered on the robot

Akustav, 2022-11-27:
The RGB sensor is not centered:

[image]

Additionally there is no FOV cone to check if the cameras location is good.

ToxaRyd, 2022-11-27:
about FOV con - I have leaved all parameters (height at least and camera plate holder angle) the same as in the old one, I was thinking to eliminate any possibilities for my team to redo calibration in their SW and waste time on it, so the camera location is good, except is that RGB camera is not centered yet...

Akustav, 2022-11-27:
Would be good to see that the current thrower to camera location is good or did that also not change?

ToxaRyd, 2022-11-28:
hard to remember, but any minor changes won't change much in that respect

ToxaRyd, 2022-11-29:
donedonedone

closed 2022-11-30 by Akustav