ut-robotics/picr22-team-snax #48: Reverse voltage protection MOSFET connected incorrectly

ReikoR, 2022-11-29:
[image]

Gate needs to be connected to GND.

markus-marandi, 2022-12-02:
done

ReikoR, 2022-12-03:
[image]

* Gate (G) needs to be connected to GND.
* Source (S) needs to be connected to +16.8.
* Bat- needs to be connected to GND.

Don't add additional labels to wires connected to power port. Power port already assigns a net name.
[image]

markus-marandi, 2022-12-03:
corrected connections

[image]

closed 2022-12-04 by ReikoR