ut-robotics/picr22-team-snax #51: DRV8243 DRVOFF incorrectly connected

ReikoR, 2022-11-29:
[image]

DRVOFF pin should not be connected to ITRIP pin.

markus-marandi, 2022-12-02:
done

ReikoR, 2022-12-03:
[image]

100k resistor needs to be connected to ITRIP pin to configure threshold voltage for current limiting.

ReikoR, 2022-12-04:
DRVOFF pin can be connected straight to GND without any resistors.
`DRV_OFF_OUT` net label can be removed. Currently all DRVOFF pins would be connected together.

[image]

markus-marandi, 2022-12-04:
[image]

closed 2022-12-05 by ReikoR