ut-robotics/picr22-team-snax #53: Use 4 layers instead of 2

ReikoR, 2022-11-29:
4 layer PCB with both inner layers being grounds is much better for signal integrity.

If there are gaps in ground (or any other reference layer), then return currents have to go around the gaps and electromagnetic fields will spread out more causing more crosstalk.

Also, if distance between signal and reference plane is bigger then fields will spread out more and cause more crosstalk.
In 4-layer PCB, distances between layers 1-2 and 3-4 are usually small (0.2mm or less).

More detailed design advice:
https://www.protoexpress.com/blog/best-high-speed-pcb-routing-practices/

Also, from JLCPCB both 2-layer and 4-layer PCBs up to 50x50mm size cost the same.

Recommended layer usage:
[image]

markus-marandi, 2022-12-02:
done

ReikoR, 2022-12-03:
Add ground polygons to inner layers.

closed 2022-12-05 by ReikoR