ut-robotics/picr22-team-snax #55: Improve component placement

ReikoR, 2022-11-30:
1. Divide PCB into two parts:
    * microcontroller and everything connected to it;
    * motor drivers and related components.
\
Something like this:
[image]

3. Place components that connect to each other close to each other.
\
ESD protection needs to be as close as possible to USB connector, otherwise ESD spikes will reach other components before ESD protection.
[image]
\
It's better to place USB connector close to microcontroller, instead of routing long tracks across the PCB.
[image]
\
Decoupling capacitors should be placed as close as possible to power supply pins of a component. Shorter connections have less inductance.
[image]

ReikoR, 2022-12-03:
1. Isolator rotated incorrectly. Microcontroller side signals are on the other side.
[image]
[image]

2. Thrower connector needs to be on the motor drivers side.
[image]

3. Place decoupling capacitors close to supply pins on the same layer as the component.
[image]

markus-marandi, 2022-12-04:
1. rotated isolator
2. moved thrower connector
3. replaced decoupling capacitors
4. replaced esd protection 
5. moved and rotated programmer connector

closed 2022-12-06 by ReikoR