ut-robotics/picr22-team-snax #57: Use wider tracks, polygons and multiple vias for higher currents

ReikoR, 2022-11-30:
[image]

0.127 mm wide track can carry about 1 A of current at 20 degrees temperature rise:
[image]

Each motor driver is limited to about 3 A in the current design.

I don't think it's possible to carry 9 A of current with 0.127 mm wide track.

Motor driver inputs and outputs need to able to carry 3 A of current.
[image]

ReikoR, 2022-12-06:
Polygon not connected to battery input pad:
[image]

markus-marandi, 2022-12-06:
done