ut-robotics/picr22-team-snax #59: Switches and power plugs covered by PC

Akustav, 2022-11-30:
The computers chassis covers the switches and sockets:

[image]

Also refer to this page for desired hole sizes:

https://ut-robotics.github.io/picr22-home/basketball_robot_guide/mechanics/design.html#_pc_power_adapter_socket

ToxaRyd, 2022-12-01:
Yep, those are bad, I decided to make them 3d printed and attached to the bottom plate!

Akustav, 2022-12-01:
The current switches can be triggered by an opponent or ball. If you dont have room I suggest putting them here: 
[image]

ToxaRyd, 2022-12-01:
Those switches are too big, putting them there will jeopardize the camera holder leg seriously (at least it does not seem reliable, and I don't like those things) But I will think on where else to put those

ToxaRyd, 2022-12-02:
Actually, I am a lazy ass, so I decided to make as you said, but the charging port remains on the holder 3d printed plate. Please check if it is ok...

maidlam, 2022-12-04:
Make the cutouts for the switch holes like this (from the guide).
If the hole is too thick, then the clips on the sides of the switches wont engage and the switch can come out.

[image]

ToxaRyd, 2022-12-04:
> Make the cutouts for the switch holes like this (from the guide). If the hole is too thick, then the clips on the sides of the switches wont engage and the switch can come out.
> 
> [image]

the hole is 4.8mm, but the part is one. meaning that on both parts manufactured the whole gonna be from on side only. So this thing will be done manualy

maidlam, 2022-12-04:
Why make them by hand?
They only need a pocket and contour operation from one side, which the CNC can cut out quickly.

ToxaRyd, 2022-12-04:
> Why make them by hand? They only need a pocket and contour operation from one side, which the CNC can cut out quickly.

The part gonna be then asymmetrical, I will need to do two walls' models. And I cannot do this cut from both sides...

maidlam, 2022-12-05:
> I don't think there gonna be any problems with reaching the other side of the plate...
The switch can reach the other side, but it wont be in place correctly, because the clips wont engage.

> I will need to do two walls' models
You can make one of them and then create a mirror body with the mirror feature.

ToxaRyd, 2022-12-05:
done

closed 2022-12-08 by ToxaRyd