ut-robotics/picr22-team-snax #64: Gap between BUK6Y10-30P drain pads

ReikoR, 2022-12-03:
[image]

Try to get rid of the gap or just use one pad for drain.

markus-marandi, 2022-12-04:
[image]

[image]

PCB editor still shows a cap, when its corrected from library and updated

ReikoR, 2022-12-04:
Which library are you using?

 It's not in the repository.

[image]

markus-marandi, 2022-12-05:
added library to repository

closed 2022-12-05 by ReikoR