ut-robotics/picr22-team-snax #67: CAM operations issues

maidlam, 2022-12-04:
Drill:

- In the Heights tab, check over the From and Offset of the operation heights
https://ut-robotics.github.io/picr22-home/basketball_robot_guide/mechanics/cam.html#_drilling

Pocket:

- Add -0,05 mm of Radial Stock to Leave, set axial to 0.
https://ut-robotics.github.io/picr22-home/basketball_robot_guide/mechanics/cam.html#_pocket

2D Contour (inner)

- Enable tabs, change the Positioning Method to "Segment", adjust the tab distance so it generates two tabs inside the motor mount holes.
This is done so that the material left in the hole doesn't fly out after cutting.
- Add -0,05 mm of Radial Stock to Leave, set axial to 0.
- Under the Linking tab, disable "Keep Tool Down", it's causing some unwanted cuts when moving between contours.

ToxaRyd, 2022-12-04:
got it!

ToxaRyd, 2022-12-05:
Almost everything was fixed, though the outer contour swears on leads, dunno how to solve that

ToxaRyd, 2022-12-05:
Actually I did fix the lead problem by specifying the "Add -0,05 mm of Radial Stock to Leave, set axial to 0" in outer contour

ToxaRyd, 2022-12-05:
So I was contacted by my teram member to make some changes into pcb fitting, and after that I got a warning of working offset 
[image]

ToxaRyd, 2022-12-05:
everything was donedonedone

closed 2022-12-08 by ToxaRyd